edenhaus/esphome-extraflame #1: Ligação ESP a Board

pedrware, 2021-05-25:
Hi,

Can you show how ESP connects to the board?

Please

edenhaus, 2021-05-25:
Hi, I'm busy this week, but I will provide you hopefully with some information by the end of next week.

I will also update the component to meet the new "external component" structure introduced in 1.18, which makes it a lot easier to use this component

pedrware, 2021-07-08:
Do you have any news?

edenhaus, 2021-07-19:
Sorry it's summer and I totally forgotten this issue.
My oven is switched off and I will active it in September. Until there you need to use the current installation in the readme file.
Probably by the end of september I will be switching to the new "external component" way

edenhaus, 2021-09-21:
I totally misunderstood your question. I will add some details soon. I will starting to prepare my oven for the winter in the next weeks

closed 2021-10-05 by pedrware