edenhaus/esphome-extraflame #11: Connecting stove with ESP8266

bachtsevanis, 2023-02-27:
Hi @vyruz1986 and @edenhaus,

Can you please help with the wiring from the stove to the ESP8266 or ESP32?

@vyruz1986 replied in another thread how to connect GND to GND and DATA to TX and RX. However, it's not clear to me since I don't have any experience with schematics. It would be nice if you could answer a few questions:

1. @vyruz1986 according to the schematic that you shared you don't connect DATA to TX and RX named pins on ESP8266. I suppose that you mean TX and RX the pins that you set in the program configuration?
2. Do you connect 5v to ESP8266 from the stove board or you are using and external power source such as USB Power adapter and you only connect GDN and DATA from the stove?
3. Is it better to use and ESP32? If so, I need to use a logic shifter to convert 5v to 3.3v. How you wire the board with that logic shifter https://www.sparkfun.com/products/12009?

edenhaus, 2023-02-27:
I use the logic shifter from your link above as in the schematic is specified and don't use the 5v from the stove as my esp32 is also doing other things and therefore I have a dedicated power supply.
An Esp8266 should be fine, but it depends if you are doing other things too.
And yes Data is connected over the logic shifter to TX/RX on the pins you define (esp32 can have multiple TX/RX)

bachtsevanis, 2023-03-02:
@edenhaus it works!!! :)

Thank you!

Have you added more sensors on maybe a way to increase / reduce power. I want to change water temperature and power based on weather conditions.